imc-trading/svlangserver #1: fileDepsInfo.get is not a function or its return value is not iterable

chrbirks, 2021-05-18:
Hi
I ran into a fatal error when trying to get svlangserver up and running on Emacs with lsp-mode. The error happens for all SystemVerilog files shortly after the file is opened in Emacs and lsp-mode tries to run the server. Maybe you have an idea to what the problem is based on the logs from Emacs below.

*svlangserver::stderr*
```
/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:292
        for (let fileDep of fileDepsInfo.get(file)) {
                                         ^

TypeError: fileDepsInfo.get is not a function or its return value is not iterable
    at SystemVerilogIndexer._incDepsRank (/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:292:42)
    at SystemVerilogIndexer._incDepsRank (/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:302:22)
    at /usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:287:22
    at Map.forEach (<anonymous>)
    at SystemVerilogIndexer._updatePkgFilesInfo (/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:284:32)
    at ChildProcess.<anonymous> (/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:198:22)
    at ChildProcess.emit (node:events:365:28)
    at emit (node:internal/child_process:920:12)
    at processTicksAndRejections (node:internal/process/task_queues:84:21)

Process svlangserver stderr finished
```

*lsp-log*
```
Command "/usr/bin/svlangserver" is present on the path.
Command "hdl_checker --lsp" is present on the path.
Command "/usr/bin/svlangserver" is present on the path.
Command "hdl_checker --lsp" is present on the path.
Found the following clients for /home/cbs/etc/example_code/systemverilog/crc32_8.sv: (server-id svlangserver, priority -1), (server-id lsp-verilog, priority -2)
The following clients were selected based on priority: (server-id svlangserver, priority -1)
This server does not support foldingRangeProvider
INFO: settings[systemverilog.includeIndexing] = **/*.{sv,svh}
INFO: settings[systemverilog.excludeIndexing] = test/**/*.{sv,svh}
INFO: settings[systemverilog.launchConfiguration] = /usr/bin/verilator -sv --lint-only -Wall
INFO: settings[systemverilog.defines] = 
INFO: settings[systemverilog.lintOnUnsaved] = true
INFO: settings[systemverilog.formatCommand] = /usr/bin/verible-verilog-format
INFO: settings[systemverilog.disableCompletionProvider] = false
INFO: settings[systemverilog.disableHoverProvider] = false
INFO: settings[systemverilog.disableSignatureHelpProvider] = false
INFO: settings[systemverilog.disableLinting] = false
INFO: Indexing 87 files ...
INFO: Loaded cached index for 87 files
INFO: Done indexing 115 files!!!
INFO: Took 770.2585819996893 milliseconds
```
*Messages*
```
LSP :: svlangserver:211705 initialized successfully in folders: (/home/cbs/etc/example_code/systemverilog)
LSP :: svlangserver has exited (exited abnormally with code 1)
Server svlangserver:211705 exited with status exit(check corresponding stderr buffer for details). Do you want to restart it? (y or n) n
```

Software versions
```
$ uname -a
Linux carbon 5.12.2-1-MANJARO #1 SMP PREEMPT Fri May 7 17:53:15 UTC 2021 x86_64 GNU/Linux
$ svlangserver -v
Version is 0.3.2
$ emacs --version
GNU Emacs 28.0.50
$ lsp-mode
20210517.213
```

*init.el*
```
  (require 'lsp-verilog)
  (custom-set-variables
   '(lsp-clients-svlangserver-launchConfiguration "/usr/bin/verilator -sv --lint-only -Wall")
   '(lsp-clients-svlangserver-formatCommand "/usr/bin/verible-verilog-format"))
  (add-hook 'verilog-mode-hook #'lsp-deferred)
```

kkanhere, 2021-05-18:
Wondering if the cache is corrupted somehow. Can you please try deleting the .emacs/.svlangserver directory in your workspace folder and then re-invoking emacs ? Do you see any errors in the *lsp-log* on the first invocation after deletion of cache?

Thanks

chrbirks, 2021-05-19:
No I get the same error after deleting .emacs. Here is the output of lsp-log (but opening a file in a folder with fewer source files than before)
```
Command "/usr/bin/svlangserver" is present on the path.
Command "hdl_checker --lsp" is present on the path.
Command "/usr/bin/svlangserver" is present on the path.
Command "hdl_checker --lsp" is present on the path.
Found the following clients for /home/cbs/tmp/sv/crc32_8.sv: (server-id svlangserver, priority -1), (server-id lsp-verilog, priority -2)
The following clients were selected based on priority: (server-id svlangserver, priority -1)
This server does not support foldingRangeProvider
INFO: settings[systemverilog.includeIndexing] = **/*.{sv,svh}
INFO: settings[systemverilog.excludeIndexing] = test/**/*.{sv,svh}
INFO: settings[systemverilog.launchConfiguration] = /usr/bin/verilator -sv --lint-only -Wall
INFO: settings[systemverilog.defines] = 
INFO: settings[systemverilog.lintOnUnsaved] = true
INFO: settings[systemverilog.formatCommand] = /usr/bin/verible-verilog-format
INFO: settings[systemverilog.disableCompletionProvider] = false
INFO: settings[systemverilog.disableHoverProvider] = false
INFO: settings[systemverilog.disableSignatureHelpProvider] = false
INFO: settings[systemverilog.disableLinting] = false
INFO: Indexing 6 files ...
INFO: Done indexing 8 files!!!
INFO: Took 333.5421319976449 milliseconds
```
I looked into if it's a specific one of my source files that is causing the error but it doesn't seem like it. I reduced my source catalog to two files that individually doesn't cause the error, but when both present it happens. I don't have them uploaded anywhere but I can paste them below in case you want to try and reproduce the error.

common_pkg.sv
```
package common_pkg;
  parameter TEST_PARAM = 12;
  int test_int = 14;

  typedef struct packed {
    logic        a;
    logic        b;
  } mytype2_t;

endpackage // common_pkg
```
testmodules.sv
```
`include "common_pkg.sv";
import common_pkg::*;

module testmodule (
  input logic                    clk_i,
  input [$clog2(TEST_PARAM)-1:0] aa_i,
  input logic                    a_i,
  output logic [1:0]             b_o
  );

  mytype2_t my;

  initial begin
    my.a = 0;
    my.b = 0;
  end

  always @(posedge clk_i) begin
    undeclared_var <= 1'b1;

    my.a <= 1'b1;
    my.b <= 1'b1;
  end
  logic undeclared_var;

  always_comb begin
    tmpa = tmpb;
    tmpb = 1;
  end
  logic tmpa;
  logic tmpb;

  
  
endmodule // testmodule
```

kkanhere, 2021-05-19:
Thanks. I will check this out on my end.

May I ask why the common_pkg.sv is being included at the top? This will cause the tool to lint that file twice.

If you exclude the common_pkg.sv from indexing (using the systemverilog.excludeIndexing setting), does that cause the error to go away?

chrbirks, 2021-05-19:
These files are just a collection of SystemVerilog files I keep to test with different language servers and Emacs configurations. I didn't write all of them myself and some might even contain syntax errors.
But from your comment it looks like it's triggered by common_pkg being included twice. If I delete the first line of testmodules.sv with `include I don't get the error anymore.

kkanhere, 2021-05-19:
That is a good confirmation that the double linting might be an issue. I will look into fixing the error. Meanwhile hopefully this workaround is ok for you.
Thanks

kkanhere, 2021-05-31:
Fixed in 0.3.4 release.

closed 2021-05-31 by kkanhere