imc-trading/svlangserver #2: Error reported as Info

chrbirks, 2021-06-13:
I was testing svlangserver with Emacs and lsp-mode and saw an error that was reported as an Info. I'm not sure if it's a problem with lsp-mode for Emacs, or svlangserver. The error/info occurs after restarting Emacs if I delete the .emacs directory with Emacs running. But the server still seem to start up fine and linting works too.

```
Command "/usr/bin/svlangserver" is present on the path.
Command "/usr/bin/svlangserver" is present on the path.
Found the following clients for /home/cbs/etc/example_code/systemverilog/copy/common_pkg.sv: (server-id svlangserver, priority -1)
The following clients were selected based on priority: (server-id svlangserver, priority -1)
This server does not support foldingRangeProvider
INFO: INFO: settings[systemverilog.includeIndexing] = copy/**/*.{v,vh,sv,svh}
INFO: INFO: settings[systemverilog.excludeIndexing] = mytest/**/*.{v,vh,sv,svh},copy/mytest/**/*.{v,vh,sv,svh}
INFO: INFO: settings[systemverilog.launchConfiguration] = /usr/bin/verilator -sv --lint-only -Wall
INFO: INFO: settings[systemverilog.defines] = 
INFO: INFO: settings[systemverilog.lintOnUnsaved] = true
INFO: INFO: settings[systemverilog.formatCommand] = /usr/bin/verible-verilog-format
INFO: INFO: settings[systemverilog.disableCompletionProvider] = false
INFO: INFO: settings[systemverilog.disableHoverProvider] = false
INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false
INFO: INFO: settings[systemverilog.disableLinting] = false
INFO: INFO: Indexing 4 files ...
INFO: Error: ENOENT: no such file or directory, open '/home/cbs/etc/example_code/systemverilog/.emacs/svlangserver/index.json'

INFO: INFO: Done indexing 4 files!!!
INFO: INFO: Took 206.20226600021124 milliseconds
```

Emacs: 28.0.50
lsp-mode: 20210612.1839
svlangserver: 0.3.4

kkanhere, 2021-06-15:
Thanks for reporting this.
Do you see this only the first time the language server is used, or on repeated invocations? I left this in to indicate that the cache could not be loaded as there is no cache file, hence it is an INFO message. But that should happen only on the first invocation. Post that the cache file should get created and this error should disappear.

chrbirks, 2021-06-16:
The info/error occurs only the first time when the .emacs directory does not exist. After the language server starts, .emacs is created again and I don't see the message next time.
So it sounds like it's intended behavior but I just wanted to check if it was a bug somewhere. I'm seeing a lot of other cryptic errors for larger projects so I'm trying to figure out if they are a result of other errors.

kkanhere, 2021-06-16:
What kind of errors?
If the defines aren't correctly provided via settings, I have seen the parser complain with errors.

chrbirks, 2021-06-16:
It is for example errors like these:
```
ERROR: Parsing failed for token `uvm_component_param_utils in file /home/cbs/etc/example_code/systemverilog/environment_packages/vip_axi4_env_pkg/src/vip_axi4_environment.svh

ERROR: TypeError: Cannot read property 'start' of undefined

ERROR: TypeError: Cannot read property 'start' of undefined
ERROR: TypeError: Cannot read property 'start' of undefined

ERROR: TypeError: Cannot read property '0' of null
ERROR: TypeError: Cannot read property '0' of null
ERROR: TypeError: Cannot read property '0' of null
```
The first one about uvm_component_param_utils is trivial since the macro is not defined anywhere in the source files, but the following does not seem to have anything to do with that file. So I'm thinking they might be a consequence of earlier errors and I should just start fixing the errors reported first.

For a larger project I get the errors in the *lsp-log* buffer you see below. The stacktrace at the end is probably just a result of trying to parse too much source code and I can probably exclude some unused code. The other errors about failed tokens are the ones I referred to as cryptic errors but I just realized they might come from trying to parse encrypted .sv files. I can't detect that from the filename so I'll have to find some other way to exclude them.
```
ERROR: Parsing failed for token `� in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `f;�Qkw�Q�ugx��j]����h�w]�U_P�j^���1��Q�&m[HN�(B�E��uaֵw�k�/��Yc��V��A0*9C@Ĩ�V ���3L�$���Y�q*��L���3GFo_YIZ���C�!��g�[���2�o���S&�y�j�4)^%ּ�A������[d�.��5�Q��- in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `�� in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token ` in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `KQ in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `O in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `dj in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `E in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

ERROR: Parsing failed for token `�� in file /mnt/storage/projects/intel/iofs-clean/bbs/design/hbm/stratix10/pac_lc/ip/hbm_bot_ctrl_ed/qii/ip/ed_synth/ed_synth_hbm_bot_ctrl_example_design/altera_abstract_uib_191/synth/altera_hbm_arch_nd_hbmc.sv

<--- Last few GCs --->

[31285:0x55ead81c5390]    61921 ms: Mark-sweep 1595.8 (1682.3) -> 1595.7 (1685.8) MB, 454.1 / 0.0 ms  allocation failure GC in old space requested
[31285:0x55ead81c5390]    62400 ms: Mark-sweep 1595.7 (1685.8) -> 1595.7 (1647.8) MB, 478.7 / 0.0 ms  last resort GC in old space requested
[31285:0x55ead81c5390]    62870 ms: Mark-sweep 1595.7 (1647.8) -> 1595.7 (1647.8) MB, 470.2 / 0.0 ms  last resort GC in old space requested

<--- JS stacktrace --->

==== JS stack trace =========================================

Security context: 0x5b8bf518fe1 <JSObject>
    1: tokenize [/usr/local/lib/node_modules/@imc-trading/svlangserver/lib/grammar_engine.js:~199] [pc=0x9c3a300e029](this=0x2994cd429f39 <GrammarEngine map = 0x2b9e9b20b3c1>,text=0x2b3542382201 <Very long string[1147553]>,initScopeStack=0x5b8bf502241 <undefined>)
    2: arguments adaptor frame: 1->2
    3: tokenize [/usr/local/lib/node_modules/@imc-trading/svlangserver/lib/svparser.js:~336] [pc=...

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory

 1: 
node::Abort() [/usr/bin/node]
 2: 
0x55ead6044011 [/usr/bin/node]
 3: 
v8::Utils::ReportOOMFailure(char const*, bool) [/usr/bin/node]
 4: 
v8::internal::V8::FatalProcessOutOfMemory(char const*, bool) [/usr/bin/node]
 5: 
v8::internal::Factory::NewUninitializedFixedArray(int) [/usr/bin/node]
 6: 
0x55ead5c30add [/usr/bin/node]
 7: 
v8::internal::Runtime_GrowArrayElements(int, v8::internal::Object**, v8::internal::Isolate*) [/usr/bin/node]
 8: 0x9c3a30040bd
```

kkanhere, 2021-06-17:
The language server definitely has trouble with verification constructs as mentioned in the README. I plan to re-write the parser to make it robust, but until then, I would recommend not indexing verification files.

chrbirks, 2021-06-17:
Yes and Verilator doesn't support UVM yet anyway so I'll exclude verification files, but the errors above are from a synthesizable example design so for now I'll also exclude the parts of the design that I know have encrypted files.
But I think I got my questions answers so I'll close this issue again.

closed 2021-06-17 by chrbirks