imc-trading/svlangserver #3: svlangserver exited with status exit when I call M-x lsp-format-buffer or use C-x h TAB to format the buffer

erusyd, 2021-08-12:
**Environment**: Ubuntu 20.04 LTS amd64, Emacs 27.1

**Problem**: When I want to format a verilog file in emacs use M-x lsp-format-buffer or use C-x h TAB, the svlangserver exited and remind me to relaunch svlangserver.

**svlangserver::stderr information as follow**:

events.js:352
      throw er; // Unhandled 'error' event
      ^

Error: write EPIPE
    at afterWriteDispatched (internal/stream_base_commons.js:156:25)
    at writeGeneric (internal/stream_base_commons.js:147:3)
    at Socket._writeGeneric (net.js:787:11)
    at Socket._write (net.js:799:8)
    at writeOrBuffer (internal/streams/writable.js:358:12)
    at Socket.Writable.write (internal/streams/writable.js:303:10)
    at Socket.Writable.end (internal/streams/writable.js:567:10)
    at Socket.end (net.js:592:31)
    at /home/syd/.nvm/versions/node/v14.17.3/lib/node_modules/@imc-trading/svlangserver/lib/svformatter.js:53:34
    at new Promise (<anonymous>)
Emitted 'error' event on Socket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:82:21) {
  errno: -32,
  code: 'EPIPE',
  syscall: 'write'
}

Process svlangserver stderr finished

kkanhere, 2021-08-13:
Thanks for reporting this. The language server uses verible-verilog-format tool for formatting. Can you confirm if you have the tool installed and in path? You can also provide the path by setting the lsp-clients-svlangserver-formatCommand variable

erusyd, 2021-08-16:
Thanks for your reply. I changed the path provided by lsp-clients-svlangserver-formatCommand  from
[image]
to
[image]
It's OK now.
I'll close the issue.

closed 2021-08-16 by erusyd