imc-trading/svlangserver #15: How to use snippets in emacs with lsp-mode?

nowislewis, 2021-12-08:
Thanks for your great work. I've configured emacs with readme.md and downloaded snippets, but I don't know how to make these snippets work.

kkanhere, 2021-12-12:
Have you tried yasnippets?

nowislewis, 2021-12-13:
Yes, I installed yasnippsets and lsp-mode. Cpp and Python work well. But it seems that the server can’t find the sv snippets, where should I put the snippets?

kkanhere, 2021-12-13:
The language server doesn't interact with the snippets. They are meant only for snippet extension such as yasnippet. Can you please try saving the snippets under ~/.emacs.d/snippets/verilog-mode and see if that works?

nowislewis, 2021-12-13:
I tried it a few minutes ago and I moved the systemverilog.json under the git repository to " ~/.emacs.d/snippets/verilog-mode/".

[image]
What really happened is that after clicking, the contents of the entire json file will be inserted into the file

kkanhere, 2021-12-13:
oh yeah. The format that yasnippet expects is different I guess. The json based snippets work with VSCode and coc-snippets (vim) I believe

nowislewis, 2021-12-13:
It seems that yas cannot work with json, I will try to rewrite  the systemverilog.json file into yas format。

closed 2021-12-13 by nowislewis