imc-trading/svlangserver #19: cant jump in windows vscode

aiit-linry, 2022-03-22:
[image]
it seems that path symbol "\" in windows is mistaken for %5C. Is this caused by svlangserver or vscode ?

kkanhere, 2022-03-25:
Can you please check the .vscode/svlangserver/index.json or .vscode/svlangserver/linter.vc file and confirm the file paths there? If those also contain '%5C' then it might be a bug with svlangserver.

kkanhere, 2022-03-30:
I am able to reproduce this on my end and it is a bug with the language server. So thanks for reporting. I have a fix and I expect it to go into the release planned in a day or two.

kkanhere, 2022-03-30:
The issue should be fixed in the latest (0.4.0) release. Kindly confirm.
Thanks

aiit-linry, 2022-04-02:
The issue of jump on **instantiated module name** is fixed, but  jump on **instantiated signal names** still exists. It is the same phenomenon that find 5C% in path.