imc-trading/svlangserver #21: svlangserver run error in coc.nvim + gvim + win11 envirement

pangchol, 2022-05-10:
When use  coc.nvim + gvim in win11 system envirement, svlangserver run error, after using "npm install -g @imc-trading/svlangserver" install svlangserver, just run svlangserver in cmd, error is:

                throw new Error('Message header must separate key and value using :');
                ^

Error: Message header must separate key and value using :
    at MessageBuffer.tryReadHeaders (D:\home\windows\bin\node_modules\@imc-trading\svlangserver\node_modules\vscode-languageserver\node_modules\vscode-jsonrpc\lib\common\messageBuffer.js:87:23)
    at StreamMessageReader.onData (D:\home\windows\bin\node_modules\@imc-trading\svlangserver\node_modules\vscode-languageserver\node_modules\vscode-jsonrpc\lib\common\messageReader.js:132:45)
    at ReadStream.<anonymous> (D:\home\windows\bin\node_modules\@imc-trading\svlangserver\node_modules\vscode-languageserver\node_modules\vscode-jsonrpc\lib\common\messageReader.js:122:18)
    at ReadStream.emit (node:events:390:28)
    at addChunk (node:internal/streams/readable:315:12)
    at readableAddChunk (node:internal/streams/readable:289:9)
    at ReadStream.Readable.push (node:internal/streams/readable:228:10)
    at TTY.onStreamRead (node:internal/stream_base_commons:199:23)

open gvim.exe and using ":CocCommand workspace.showOutput" to see log, show is:

[Error  - 下午11:06:37] Starting client failed 
Launching server "languageserver.svlangserver" using command svlangserver failed.
[Error  - 下午11:06:37] spawn svlangserver ENOENT

suggest fix it.

kkanhere, 2022-05-22:
Are any other language servers working? Because from the error you have posted, it looks like the fundamental vscode json rpc is broken. That part is not specific to svlangserver. Please check if your system supports vscode-jsonrpc-6.0.0

pangchol, 2022-05-22:
In my win11 envirement, the coc.nvim with clangd language server is working fine. In addition, my some ohter coc.nvim's extention also no product error so far.

kkanhere, 2022-05-22:
Can you please post the versions of coc.nvim and vscode? My guess is that those don't support version 6.0.0 for the vscode-jsonrpc

pangchol, 2022-05-22:
here is CocInfo print:
```
## versions

vim version: VIM - Vi IMproved 8.2 8023000
node version: v16.13.2
coc.nvim version: 0.0.80-b7cd84e 2022-03-17 17:24:36 +0800
coc.nvim directory: /home/zhougq/.home/.vim/plug/coc.nvim
term: tmux
platform: linux

## Log of coc.nvim

```

pangchol, 2022-05-22:
when useing svlangserver, CocInfo print is:
```
## versions

vim version: VIM - Vi IMproved 8.2 8023000
node version: v16.13.2
coc.nvim version: 0.0.80-b7cd84e 2022-03-17 17:24:36 +0800
coc.nvim directory: D:\home\.vim\plug\coc.nvim
term: undefined
platform: win32

## Log of coc.nvim

2022-05-22T23:41:41.209 INFO (pid:4580) [services] - registered service "languageserver.clangd"
2022-05-22T23:41:41.211 INFO (pid:4580) [services] - registered service "languageserver.svlangserver"
2022-05-22T23:41:41.214 INFO (pid:4580) [services] - svlangserver state change: stopped => starting
2022-05-22T23:41:41.257 INFO (pid:4580) [services] - registered service "highlight"
2022-05-22T23:41:41.312 INFO (pid:4580) [plugin] - coc.nvim initialized with node: v16.13.2 after 202ms
2022-05-22T23:41:41.317 INFO (pid:4580) [services] - svlangserver state change: starting => stopped
2022-05-22T23:41:41.330 ERROR (pid:4580) [services] - Server languageserver.svlangserver failed to start: Launching server "languageserver.svlangserver" using command svlangserver failed.
2022-05-22T23:41:41.330 ERROR (pid:4580) [language-client-index] - Error: spawn svlangserver ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:282:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'spawn svlangserver',
  path: 'svlangserver',
  spawnargs: []
}
2022-05-22T23:41:48.403 INFO (pid:4580) [attach] - receive notification: showInfo []

```

pangchol, 2022-05-22:
I use npm install -g @imc-trading/svlangserver to install the svlangserver in windows, then try open a verilog file， then the CodInfo is:
````
## versions

vim version: VIM - Vi IMproved 8.2 8023000
node version: v16.13.2
coc.nvim version: 0.0.80-b7cd84e 2022-03-17 17:24:36 +0800
coc.nvim directory: D:\home\.vim\plug\coc.nvim
term: undefined
platform: win32

## Log of coc.nvim

2022-05-22T23:50:15.952 INFO (pid:11984) [services] - registered service "languageserver.clangd"
2022-05-22T23:50:15.954 INFO (pid:11984) [services] - registered service "languageserver.svlangserver"
2022-05-22T23:50:15.956 INFO (pid:11984) [services] - svlangserver state change: stopped => starting
2022-05-22T23:50:16.000 INFO (pid:11984) [services] - registered service "highlight"
2022-05-22T23:50:16.051 INFO (pid:11984) [plugin] - coc.nvim initialized with node: v16.13.2 after 193ms
2022-05-22T23:50:16.055 INFO (pid:11984) [services] - svlangserver state change: starting => stopped
2022-05-22T23:50:16.066 ERROR (pid:11984) [services] - Server languageserver.svlangserver failed to start: Launching server "languageserver.svlangserver" using command svlangserver failed.
2022-05-22T23:50:16.067 ERROR (pid:11984) [language-client-index] - Error: spawn svlangserver ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:282:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'spawn svlangserver',
  path: 'svlangserver',
  spawnargs: []
}
2022-05-22T23:50:19.529 INFO (pid:11984) [attach] - receive notification: showInfo []
````

kkanhere, 2022-05-22:
Can you please try building from source. I think the below steps should be done
1. Clone the repo
2. cd svlangserver && npm install
3. Update your .vim/coc-settings.json to point to <your path>/svlangserver/bin/main.js

If this works, then kindly post the contents of package-lock.json

pangchol, 2022-05-23:
I try it:
1. git clone https://github.com/imc-trading/svlangserver.git (clone to the D:/home/windows/coc/svlangserver dir)
2. cd to repo dir and npm install, it build sucess
3. set coc-setings.json link this:
```
        "svlangserver": {
            "command": "D:/home/windows/coc/svlangserver/bin/main.js",
            "filetypes": ["systemverilog", "verilog"],
            "settings": {
                "systemverilog.includeIndexing": ["**/*.{v,sv,svh}"],
                "systemverilog.excludeIndexing": ["test/**/*.sv*"],
                "systemverilog.defines" : [],
                "systemverilog.launchConfiguration": "verilator -sv -Wall --lint-only",
                "systemverilog.formatCommand": "verible-verilog-format"
            }
        }
    }
```
unfortunately, when open .v file:
CocInfo is:
```
## versions

vim version: VIM - Vi IMproved 8.2 8023000
node version: v16.13.2
coc.nvim version: 0.0.80-b7cd84e 2022-03-17 17:24:36 +0800
coc.nvim directory: D:\home\.vim\plug\coc.nvim
term: undefined
platform: win32

## Log of coc.nvim

2022-05-23T18:08:01.987 INFO (pid:9196) [services] - registered service "languageserver.clangd"
2022-05-23T18:08:01.988 INFO (pid:9196) [services] - registered service "languageserver.svlangserver"
2022-05-23T18:08:01.991 INFO (pid:9196) [services] - svlangserver state change: stopped => starting
2022-05-23T18:08:02.035 INFO (pid:9196) [services] - registered service "highlight"
2022-05-23T18:08:02.084 INFO (pid:9196) [plugin] - coc.nvim initialized with node: v16.13.2 after 199ms
2022-05-23T18:08:02.087 INFO (pid:9196) [services] - svlangserver state change: starting => stopped
2022-05-23T18:08:02.099 ERROR (pid:9196) [services] - Server languageserver.svlangserver failed to start: Error: spawn UNKNOWN
    at ChildProcess.spawn (node:internal/child_process:412:11)
    at Object.spawn (node:child_process:698:9)
    at D:\home\.vim\plug\coc.nvim\build\index.js:54060:63 {
  errno: -4094,
  code: 'UNKNOWN',
  syscall: 'spawn'
}
2022-05-23T18:08:06.050 INFO (pid:9196) [attach] - receive notification: showInfo []

```

pangchol, 2022-05-24:
the path is ok when use forward slash, if path no found, the CocInfo print other error message. by the way , backslashes is also test, print the same error as "forward slash".

kkanhere, 2022-05-24:
Since the errors in the two cases are different, there is definitely something different between those two setups. Hence suspecting the path (based on the error message it is probably a good guess).

Unfortunately I don't have access to win11 system and hence can't test this myself.

kkanhere, 2022-05-26:
I was able to reproduce the error on a win10 machine. I was able to get past it by using the below coc-settings.json
```json
{
    "languageserver": {
        "svlangserver": {
            "module": "GLOBAL_NODE_MODULES_PATH/@imc-trading/svlangserver/bin/main.js",
            "args": ["--node-ipc"],
            "filetypes": ["systemverilog"],
            "settings": {
                ...
            }
        }
    }
}
```
the GLOBAL_NODE_MDULES_PATH should point to folder where node modules get globally installed.