imc-trading/svlangserver #27: do not create .vim folder and index files.

icymankr, 2022-06-17:
After update svlangserver with 'npm install -g @imc-trading/svlangserver' recently, it does not create the folder .vim in workspace. And the function of svlangserver is not working also. But I copied .vim folder from old workspace to new workspace without the folder, it is working.
I'm using coc.nvim on ubuntu.
The .vim folder found under my old workspace is created on last May.

kkanhere, 2022-06-18:
Thanks for filing the issue.
Can you please post the svlangserver output from `CocCommand workspace.showOutput`?
Because there is explicit logic in the language server to create folders recursively when needed. Maybe it is a permissions issue?

icymankr, 2022-06-19:
> Thanks for filing the issue. Can you please post the svlangserver output from `CocCommand workspace.showOutput`? Because there is explicit logic in the language server to create folders recursively when needed. Maybe it is a permissions issue?

Thanks for your reply.
As below, I don't find any suspicious message. 

            INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,sv,svh}
            INFO: INFO: settings[systemverilog.mustIncludeIndexing] = 
            INFO: INFO: settings[systemverilog.libraryIndexing] = 
            INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*
            INFO: INFO: settings[systemverilog.linter] = verilator
            INFO: INFO: settings[systemverilog.launchConfiguration] = verilator --lint-only -Wall
            INFO: INFO: settings[systemverilog.defines] = 
            INFO: INFO: settings[systemverilog.lintOnUnsaved] = true
            INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
            INFO: INFO: settings[systemverilog.formatCommand] = /home/icyman/Downloads/install/verible/bin/verible-verilog-format
            INFO: INFO: settings[systemverilog.disableCompletionProvider] = false
            INFO: INFO: settings[systemverilog.disableHoverProvider] = false
            INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false
            INFO: INFO: settings[systemverilog.disableLinting] = false
            INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,sv,svh}
            INFO: INFO: settings[systemverilog.mustIncludeIndexing] = 
            INFO: INFO: settings[systemverilog.libraryIndexing] = 
            INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*
            INFO: INFO: settings[systemverilog.linter] = verilator
            INFO: INFO: settings[systemverilog.launchConfiguration] = verilator --lint-only -Wall
            INFO: INFO: settings[systemverilog.defines] = 
            INFO: INFO: settings[systemverilog.lintOnUnsaved] = true
            INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
            INFO: INFO: settings[systemverilog.formatCommand] = /home/icyman/Downloads/install/verible/bin/verible-verilog-format
            INFO: INFO: settings[systemverilog.disableCompletionProvider] = false
            INFO: INFO: settings[systemverilog.disableHoverProvider] = false
            INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false
            INFO: INFO: settings[systemverilog.disableLinting] = false
            [Error  - 8:31:38 AM] ERROR: Timeout trying to index document file:///home/icyman/project/kddx/tests.sv

kkanhere, 2022-06-20:
The language server doesn't really support verification constructs right now. If you exclude the tests.sv file, does it fix the issue?

closed 2023-02-14 by kkanhere