imc-trading/svlangserver #31: Project configuration file

qarlosalberto, 2022-06-23:
It would be nice to support a project configuration file instead of to search in the workspace. Similar to other HDL LS. For example: https://github.com/VHDL-LS/rust_hdl_vscode

```
# File names are either absolute or relative to the parent folder of the
# vhdl_ls.toml file and supports glob-style patterns.
[libraries]
lib2.files = [
  'pkg2.vhd',
  'src/**/*.vhd',
]
lib1.files = [
  'pkg1.vhd',
  'tb_ent.vhd',
]
```

kkanhere, 2022-07-01:
Thanks for the suggestion.
What benefits does this approach offer? Even if a project config file was supported, a workspace specific file would still be needed to point to such a configuration file.

qarlosalberto, 2022-07-01:
@kkanhere could you send me a email to carlosruiznaranjo@gmail.com or we can chat in Gitter. I would like to discuss about this and other stuffs.

qarlosalberto, 2022-07-01:
I would like to talk about a possible integration in TerosHDL (https://terostechnology.github.io/terosHDLdoc/index.html)