imc-trading/svlangserver #34: Failed to get definition of macro in .svh file

walwan, 2022-09-30:
Hi:
    I encountered this problem in a project organized like this

> -- code\\
> -------- sys_def.svh
> -------- verilog\\
> ---------------- if_stage.sv
> ---------------- id_stage.sv

If I try to find the definition of a module in if_stage.sv from id_stage.sv, it can find it. However, if I try to find a definition of a macro in sys_def.svh from either if_stage.sv or id_stage.sv, it can't find it.

My environment is

> vim version: NVIM v0.7.2
node version: v18.8.0
coc.nvim version: 0.0.82-83aa3733 2022-09-29 17:42:51 +0800
coc.nvim directory: ~/.local/share/nvim/plugged/coc.nvim
term: tmux
platform: linux
svlangserver: 0.4.1/latest master (I tried both)

Here is a small example project having this problem: [poc.tar.gz](https://github.com/imc-trading/svlangserver/files/9686175/poc.tar.gz). I noticed one error message from the communication between coc.nvim and the language server:

> [Error - 12:23:11 PM] ERROR: Parsing failed for token `XLEN in file ./poc/verilog/if_stage.sv

Thanks,
walwan

henry-hsieh, 2022-10-09:
You should add `` `include "sys_defs.svh"`` in your `if_stage.sv` file, or the indexer won't find where the definition is.
Adding definition file in the source file also fixes the error that Verilator can't find your `` `XLEN`` macro.

Before:
[image]
After:
[image]

walwan, 2022-10-09:
Thank u very much.