imc-trading/svlangserver #35: ERROR: TypeError: Cannot read properties of undefined (reading 'symbolsInfo')

qian-gu, 2022-11-19:
I have install svlangserver with `npm install -g @imc-trading/svlangserver` and add it to my `coc-setting.json` file:

```
{
    "languageserver": {
        "svlangserver": {
            "command": "svlangserver",
            "filetypes": ["verilog", "systemverilog", "verilog_systemverilog"],
            "settings": {
                "systemverilog.includeIndexing": ["**/*.{v,vh,sv,svh}"],
                "systemverilog.excludeIndexing": ["test/**/*.sv*"],
                "systemverilog.defines" : [],
                "systemverilog.launchConfiguration": "verilator -sv -Wall --lint-only",
                "systemverilog.formatCommand": "verible-verilog-format"
            }
        }
    }
}
```

The server print `Version is 0.4.1` with `svlangserver --version`, but doesn't work wit coc.nvim, `:cocCommand workspace.showOutput` gives the error information:

```
INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,vh,sv,svh}                                                                                                                                                                     
 INFO: INFO: settings[systemverilog.mustIncludeIndexing] =                                           
 INFO: INFO: settings[systemverilog.libraryIndexing] =                                               
 INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*                                 
 INFO: INFO: settings[systemverilog.linter] = verilator                                              
 INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv -Wall --lint-only           
 INFO: INFO: settings[systemverilog.defines] =                                                       
 INFO: INFO: settings[systemverilog.lintOnUnsaved] = true                                            
 INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
 INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format                          
 INFO: INFO: settings[systemverilog.disableCompletionProvider] = false                               
 INFO: INFO: settings[systemverilog.disableHoverProvider] = false                                    
 INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false                            
 INFO: INFO: settings[systemverilog.disableLinting] = false                                          
 INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,vh,sv,svh}                            
 INFO: INFO: settings[systemverilog.mustIncludeIndexing] =                                           
 INFO: INFO: settings[systemverilog.libraryIndexing] =                                               
 INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*                                 
 INFO: INFO: settings[systemverilog.linter] = verilator                                              
 INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv -Wall --lint-only           
 INFO: INFO: settings[systemverilog.defines] =                                                       
 INFO: INFO: settings[systemverilog.lintOnUnsaved] = true                                            
 INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
 INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format                          
 INFO: INFO: settings[systemverilog.disableCompletionProvider] = false                               
 INFO: INFO: settings[systemverilog.disableHoverProvider] = false                                    
 INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false                            
 INFO: INFO: settings[systemverilog.disableLinting] = false                                          
 [Error - 12:53:37 AM] ERROR: TypeError: Cannot read properties of undefined (reading 'symbolsInfo') 
```

and it seems everthing is ok with `:cocInfo`:

```
## versions                                                                                         
                                                                                                     
 vim version: VIM - Vi IMproved 8.2 8024919                                                          
 node version: v18.10.0                                                                              
 coc.nvim version: 0.0.82-a7c0900f 2022-11-08 23:10:03 +0800                                         
 coc.nvim directory: /home/qian/.vim/bundle/coc.nvim                                                 
 term: dumb                                                                                          
 platform: linux                                                                                     
                                                                                                     
 ## Log of coc.nvim                                                                                  
                                                                                                     
 2022-11-20T00:53:20.815 INFO (pid:387370) [services] - LanguageClient svlangserver state change: stopped => starting
 2022-11-20T00:53:20.873 INFO (pid:387370) [plugin] - coc.nvim initialized with node: v18.10.0 after 82ms
 2022-11-20T00:53:20.874 INFO (pid:387370) [language-client-index] - Language server "languageserver.svlangserver" started with 387404
 2022-11-20T00:53:20.935 INFO (pid:387370) [services] - LanguageClient svlangserver state change: starting => running
 2022-11-20T00:53:20.938 INFO (pid:387370) [services] - service languageserver.svlangserver started  
 2022-11-20T00:53:32.588 INFO (pid:387370) [attach] - receive notification: runCommand [ 'workspace.showOutput' ]
 2022-11-20T00:53:34.434 WARN (pid:387370) [plugin] - Slow action "runCommand" cost 1846ms           
 2022-11-20T00:55:51.559 INFO (pid:387370) [attach] - receive notification: checkJsonExtension []    
 2022-11-20T00:55:51.602 INFO (pid:387370) [services] - LanguageClient Json language server state change: stopped => starting
 2022-11-20T00:55:51.611 INFO (pid:387370) [language-client-index] - Language server "json" started with 387552
 2022-11-20T00:55:51.662 INFO (pid:387370) [services] - LanguageClient Json language server state change: starting => running
 2022-11-20T00:55:51.666 INFO (pid:387370) [services] - service json started                         
 2022-11-20T00:59:14.090 INFO (pid:387370) [attach] - receive notification: showInfo []
```

I'm new to coc.nvim and node/npm , so I can't trace more, but I'd like to provide more information if needed.

kkanhere, 2022-12-09:
What is not working?
Do you see a index.json under .vim/svlangserver directory? does it contain anything?

qian-gu, 2022-12-09:
for a simple test module:

```systemverilog
module test (
  input  logic din
  // output logic dout
  // port list
);
  // module body
  always @(*) begin
    dout = din;
  end

endmodule : test
```

It is expected to show some lint error message for undefined varibale 'dout', but there are not any hint or information in vim, it seems that svlangserver doesn't work.

I'm sure that verilator is ok when call it from the terminal:

```terminal
verilator --lint-only test.sv
```
and I got following messages:

```text
%Warning-NULLPORT: test.sv:5:1: Null port on module (perhaps extraneous comma)
    5 | );
      | ^
                   ... For warning description see https://verilator.org/warn/NULLPORT?v=4.226
                   ... Use "/* verilator lint_off NULLPORT */" and lint_on around source to disable this message.
%Error: test.sv:8:5: Can't find definition of variable: 'dout'
    8 |     dout = din;
      |     ^~~~
%Error: Exiting due to 1 error(s), 1 warning(s)
```

There are no `./vim/svlangserver` directory, my svlangserve is located in `/home/qian/.nvm/versions/node/v18.10.0/lib/node_modules/@imc-trading/svlangserver`, and there are no `index.json` file under the directory.

kkanhere, 2022-12-09:
I am unable to reproduce this on my end as I tried your settings and test.sv file and it correctly gives linting errors to me. I wonder if your workspace is correctly configured - i.e. is coc.nvim detecting workspace correctly?

qian-gu, 2022-12-10:
my test project directory is as blew:

```
.
├── .git
├── rtl
│   ├── counter_pkg.sv
│   ├── counter.sv
│   ├── hello.sv
│   └── top.sv
└── .vim
    └── coc-settings.json
```

`:cocCommand workspace.workspaceFolders` output the project root directory correctly, and `:echo coc#util#root_patterns()` output 

```
{'global': ['.git', '.hg', '.projections.json'], 'buffer': ['.git', '.env'], 'server': []}
```

It seems everything is okay. 

I have also change filename to 'counter.sv` to avoid excludeIndexing, after adding `.vim/coc-setting.json` manually, nothing is generated in `.vim/` folder and I can't see any lint message still.

kkanhere, 2022-12-11:
Can you please try emptying out the excludingIndex ? (i.e. "systemverilog.excludeIndexing": [])
I am wondering if there is any bug where excludingIndex doesn't find a file the indexing mechanism breaks.

qian-gu, 2022-12-12:
The problem is solved by empty the excludingIndex, I'd like to upload my test project if you want debug with it :D

Thanks for your great job! Another problem is that I have no idea to use verible-verilog-formmat with svlangserver in vim, there is a line for it in the example configuration file, i.e., `systemverilog.formatCommand: verible-verilog-format`, but following custom command deosn't work:

```vim
command! SvFormat call CocRequest("svlangserver", 'workspace/executeCommand', {'command': 'systemverilog.formmatCommand', 'arguments': [input('--port_declarations_alignment: ', <range>      == 0 ? "" : expand("<cword>")), '--inplace=true']})
```

kkanhere, 2022-12-13:
I am able to reproduce it on my end now so I think I have everything for debugging this. Thanks

For the formatting I am able to use coc-format command. I have mapped the coc-format and coc-format-selected commands, and use those key bindings to invoke formatting.

qian-gu, 2022-12-14:
Thanks for your kindly help, I am able to use the format command now :D