imc-trading/svlangserver #38: Can't lint and format

kdurant, 2023-01-23:
I use svlangserver by coc.nvim and config like below.
Verilator and verible-verilog-format can run in terminal.
```lua
{
    "languageserver": {
        "svlangserver": {
            "command": "svlangserver",
            "filetypes": ["systemverilog"],
            "settings": {
                "systemverilog.includeIndexing": ["**/*.{sv,svh}"],
                "systemverilog.excludeIndexing": ["test/**/*.sv*"],
                "systemverilog.defines" : [],
                "systemverilog.launchConfiguration": "verilator -sv -Wall --lint-only",
                "systemverilog.formatCommand": "verible-verilog-format"
            }
        }
    }
}
```

kkanhere, 2023-02-10:
What do the logs report? You should be able to get those using the command `CocCommand workspace.showOutput` and then selecting svlangserver

kdurant, 2023-02-11:
```
INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,sv,svh}
INFO: INFO: settings[systemverilog.mustIncludeIndexing] = 
INFO: INFO: settings[systemverilog.libraryIndexing] = 
INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*
INFO: INFO: settings[systemverilog.linter] = verilator
INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv -Wall --lint-only
INFO: INFO: settings[systemverilog.defines] = 
INFO: INFO: settings[systemverilog.lintOnUnsaved] = true
INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format
INFO: INFO: settings[systemverilog.disableCompletionProvider] = false
INFO: INFO: settings[systemverilog.disableHoverProvider] = false
INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false
INFO: INFO: settings[systemverilog.disableLinting] = false
INFO: INFO: settings[systemverilog.includeIndexing] = **/*.{v,sv,svh}
INFO: INFO: settings[systemverilog.mustIncludeIndexing] = 
INFO: INFO: settings[systemverilog.libraryIndexing] = 
INFO: INFO: settings[systemverilog.excludeIndexing] = test/**/*.sv*
INFO: INFO: settings[systemverilog.linter] = verilator
INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv -Wall --lint-only
INFO: INFO: settings[systemverilog.defines] = 
INFO: INFO: settings[systemverilog.lintOnUnsaved] = true
INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i
INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format
INFO: INFO: settings[systemverilog.disableCompletionProvider] = false
INFO: INFO: settings[systemverilog.disableHoverProvider] = false
INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false
INFO: INFO: settings[systemverilog.disableLinting] = false
[Error - 22:46:44.440] ERROR: TypeError: Cannot read properties of undefined (reading 'symbolsInfo')
[Error - 22:46:47.440] ERROR: TypeError: Cannot read properties of undefined (reading 'symbolsInfo')
[Error - 22:47:28.662] ERROR: TypeError: Cannot read properties of undefined (reading 'symbolsInfo')
[Error - 22:47:42.096] ERROR: Timeout trying to index document file:///home/wj/tmp/verilog/abc.v
[Error - 22:47:44.475] ERROR: Timeout trying to index document file:///home/wj/tmp/verilog/abc.v
[Error - 22:47:47.477] ERROR: Timeout trying to index document file:///home/wj/tmp/verilog/abc.v
```

kkanhere, 2023-02-14:
Do you have a .svlangserver folder in your workspace? Does it have a file index.json?
Is the excludeIndexing pattern correct? If it doesn't match any files, then you should empty it out.
Does abc.v have any verification constructs? The language server doesn't understand verification constructs yet.

kdurant, 2023-02-14:
I don't have a .svlangserver folder in your workspace. I can't see any document about this.
```bash
wj@wj:~/tmp/verilog$ ls -al
drwxrwxr-x   - wj 11 Feb 22:44 .git
.rw-rw-r-- 199 wj 14 Feb 22:34 abc.v

```

```verilog
// abc.v
`timescale  1 ns/1 ps

module abc
(
    input                       clk,
    inpu                       rst,
);

reg abc = 0;
    always @ (posedge clk)
    begi
        abc <= 1;
    end

endmodule
```

kdurant, 2023-02-15:
```vim
    command! SvBuildIndex call CocRequest("svlangserver", 'workspace/executeCommand', {'command': 'systemverilog.build_index'})
```
I must run this command

closed 2023-02-15 by kdurant