imc-trading/svlangserver #41: Single file linting

dylanpjx, 2023-06-14:
Hi, I’m using the LSP with nvim. Under server_configurations on nvim_lspconfig, it states that there is single file support enabled by default.

However, I don’t get linting on files unless in a git repository. Is it possible to enable linting globally and how would I go about achieving this?

kkanhere, 2023-06-17:
The language server's functionality is tied to a workspace not a repo. So I would recommend checking if your LSP engine is correctly detecting a workspace and informing the language server so. Would suggest checking the deubg info from your LSP for the language server. The language server should log this info including the files it is indexing etc.

dylanpjx, 2023-06-23:
The workspace is determined by LSP's root_dir option right? Is it possible to lint individual files outside of workspaces?

kkanhere, 2023-06-29:
Unfortunately no such support exists for this language server right now. I have added it to the TBD list. But pull requests always appreciated.
Thanks