imc-trading/svlangserver #42: Error to go to definition of `define macro` in *.vh

beyond-fu, 2023-08-26:
Here is my file organization:
[image]
Going to definition of variables either in current file or across files works well. But going to definition of `define macro in params.vh` crashes.  When I press `gd(goto definition)` on `ADDR_WIDTH`, here are the error messages.
[image]
All the error messages are in the above picture. I have no idea what's wrong. Can anyone help me?

```
My Environment:
neovim v0.9.1
X86_64 Ubuntu 22.04
```
```
My nvim-lspconfig config:
I'm using LazyVim.
```
[image]

Thanks! This LSP is great!

kkanhere, 2023-09-05:
Interesting. Can you please create a small testcase that I can use to reproduce the bug on my end. Thanks

beyond-fu, 2023-09-05:
You mean a verilog project or an LSP testcase?

kkanhere, 2023-09-06:
I would need your editor details, settings, LSP settings, the verilog files. Basically everything to recreate this on my end.

beyond-fu, 2023-09-07:
OK, I will deal with it sometime. Maybe I can put all things on my github repo? And there is a very strange bug, I will open a new issue sometime.