imc-trading/svlangserver #43: Add support to expand `define macro and parameters` in hover window

beyond-fu, 2023-08-26:
I think it will be perfect if the `define macro and parameters` can be expanded in hover window. It is not so intuitive for current hover if having a macro or parameter. Is it possible?

Thanks.

kkanhere, 2023-09-05:
By expand do you mean preprocess it and fill in the tokens? Or do you mean show the definition of the macro? Because latter is what it should be doing right now. The former doesn't seem like a job for language server.

beyond-fu, 2023-09-05:
Yes, I mean preprocess the `define macro and parameter` and show the value in the hover window. And it's better if the macros and parameters can be processed recursively and displayed correctly. Isn't that what LSP is supposed to do? I don't know about the details of LSP protocol. But `Clangd` can expand `define macro` in C/C++. Thanks for your reply!

beyond-fu, 2023-09-05:
I think you can extend the functionality of this plugin beyond the LSP functionality itself.

kkanhere, 2023-09-06:
Added it to the todo list but fastest way to get the feature would be to contribute a pull request. Thanks

beyond-fu, 2023-09-07:
Unfortunately, I don't know `TypeScript` programming and the details of LSP. Thanks!