imc-trading/svlangserver #44: Need to support relative imports

Johnlon, 2023-08-27:
I have a lib of models in a separate tree of code; for example.
```
c:\mylib\therom\rom.v
c:\mylib\74574\hct74574.v
```

In icarus I include "-I c:\mylib\" and this allows an import syntax like ...
```
 `include "therom/rom.v"
```

Whilst this works fine in icarus, this does not scan in the VSCode plugin.
```
"systemverilog.includeIndexing": [        
        "**/*.v",
        "C:\\mylib"
    ],
```

Also this does not work...

```
"systemverilog.includeIndexing": [        
        "**/*.v",
        "C:\\mylib\\**\*.v"  
    ],
```

The problem is the scan results don't remember their relative path. So for the second example above this DOES work ...
```
 `include "rom.v"
```

... but that's no use to me as it doesn't work readily in icarus.

---

Also ... when using the second approach then I can't navigate to symbol in the import but that's academic as the scanner doesn't work the way icarus does.

kkanhere, 2023-09-05:
The language server only works in a workspace context. It is unable to manage files outside of a workspace.

Johnlon, 2023-09-05:
I don't believe file relative works at all regardless of location

kkanhere, 2023-09-06:
Can you clarify because I am confused what is not working here. Are you saying file relative includes don't work? Or the scanning doesn't work. Because the latter works only in workspace context so I believe absolute paths won't work. And If scanning failed, then includes will fail no matter what as the language server won't know about the file.

Johnlon, 2023-09-06:
My experience was that if I create a dir structure in the current working
directory and put a model in a sub dir eg. foo/sub.v then create a model in
the parent dir that has an include for "foo/sub.v" then that didn't work.

Actually I had a situation where a sibling child die was doing the include
"../foo/sub.v"

But both are valid test cases.