imc-trading/svlangserver #45: Linting with Verilator 5: SystemVerilog library files incorrectly processed as Verilog files

aumouvantsillage, 2023-08-28:
I am using svlangserver in Neovim with the default settings.
The linter is Verilator 5.015.

In my projects, the linter emits this error message for all `import`s:

> Importing from missing package 'P'

However, the corresponding package files are all listed correctly in the `linter.vc` file generated by svlangserver.

I have noticed that the errors disappear if I edit the file `linter.vc` and I remove the `-v` switch before the library file names.
In fact, according the Verilator man page, this option is documented as:

> -v <filename>              Verilog library

I suspect that Verilator 5 does not support SystemVerilog for files introduced with the `-v` option.
Just listing the file names alone seems to work.

kkanhere, 2023-09-05:
Thanks for filing an issue.Verilator linting is broken at the moment and I plan to fix it in the next release.

jagjordi, 2024-05-18:
Is there any plan of fixing this? What would be needed? I could give it a try!

kkanhere, 2024-05-18:
It is definitely in the TODO list.
An immediate simple fix would be to just revert the logic that was changed around linting in this release as the previous release had better linting functionality.
In the long run the idea should be to create a vc file per file being edited with precisely its dependencies.