imc-trading/svlangserver #48: Go to Definition support for duplicate module names.

haydnevans1, 2024-05-16:
Hi, really love the extension! I work in a codebase where we often have multiple modules of the same name. The current functionality of the extension seems to just pick the first module it finds of that name. Is it possible to get a list of modules that match the name when using go to Definition?

Thanks.

kkanhere, 2024-05-18:
Yep. The language server picks the first match. Since the protocol does support providing multiple targets, I believe this can be supported. Added todo list. Also note that pull requests are always welcome.

haydnevans1, 2024-05-20:
Ok great! Good to know it is very possible, I may make a pull request and give it a go over the weekend.