imc-trading/svlangserver #52: Server crashes when specifying libraryIndexing

jagjordi, 2024-11-12:
I am trying to get the LSP to include a library contains a package that I import in my SV files. however when I specify it in the `libraryIndexing` option the server crashes. this is the log:
```
[ERROR][2024-11-12 11:28:52] .../vim/lsp/rpc.lua:770    "rpc"   "/sbin/svlangserver"    "stderr"        "/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:395\n        for (let incdir of [...new Set(this._srcFiles.map(file => path.dirname(file))), ...new Set(this._srcFiles.map(file => path.dirname(file)))]) {\n                                                      ^\n\nTypeError: Cannot read properties of undefined (reading 'map')\n    at SystemVerilogIndexer._generateVerilatorOptionsFile (/usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:395:55)\n    at /usr/lib/node_modules/@imc-trading/svlangserver/lib/svindexer.js:76:26\n    at f (/usr/lib/node_modules/@imc-trading/svlangserver/node_modules/once/once.js:25:25)\n    at Glob.<anonymous> (/usr/lib/node_modules/@imc-trading/svlangserver/node_modules/glob/glob.js:148:7)\n    at Glob.emit (node:events:507:28)\n    at Glob._finish (/usr/lib/node_modules/@imc-trading/svlangserver/node_modules/glob/glob.js:194:8)\n    at next (/usr/lib/node_modules/@imc-trading/svlangserver/node_modules/glob/glob.js:213:12)\n    at /usr/lib/node_modules/@imc-trading/svlangserver/node_modules/glob/glob.js:245:9\n    at /usr/lib/node_modules/@imc-trading/svlangserver/node_modules/fs.realpath/index.js:37:7\n    at LOOP (node:fs:2839:14)\n\nNode.js v23.1.0\n"
~
```
Any idea why this happens? Am I doing something wrong? Or is it a bug?

kkanhere, 2024-12-16:
Please provide the complete config and details on ide being used, versions etc.

henry-hsieh, 2025-08-20:
It seems when the `_generateVerilorOptionsFile()` is called by `setLibraries()`, the `this._srcFiles` became undefined. Therefore, the `map()` will be reported as undefined property.

My temporary solution is adding the guard before `this._srcFiles` is referenced:
```typescript
// Guard
if (this._srcFiles == undefined) {
    this._srcFiles = []
}

for (let incdir of [...new Set(this._srcFiles.map(file => path.dirname(file))), ...new Set(this._srcFiles.map(file => path.dirname(file)))]) {
    this._optionsFileContent.push('+incdir+' + incdir);
}
```

I also have a small question, why `...new Set(this._srcFiles.map(file => path.dirname(file)))` is called twice? This will cause option file having two identical `+incdir+/path/to/index/dir`.

henry-hsieh, 2025-11-16:
I have done some test on this. Here is my log in Neovim:
```
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] onInitialized"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] onDidChangeConfiguration"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] updateSettings"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.includeIndexing] = src/**/*.{v,h}"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.mustIncludeIndexing] = "
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.libraryIndexing] = common/**/*.{v,h}"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.excludeIndexing] = **/test/**"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.linter] = verilator"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv --lint-only --Wall"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.defines] = "
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.lintOnUnsaved] = true"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableCompletionProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableHoverProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableLinting] = false"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] !settingsInitialized"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] setLibraries()"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] onInitialized to updateSettings"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] updateSettings"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.includeIndexing] = src/**/*.{v,h}"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.mustIncludeIndexing] = "
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.libraryIndexing] = common/**/*.{v,h}"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.excludeIndexing] = **/test/**"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.linter] = verilator"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.launchConfiguration] = verilator -sv --lint-only --Wall"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.defines] = "
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.lintOnUnsaved] = true"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.linterWhitelist] = /Unsupported: Interfaced port on top level module/i,/Parent instance's interface is not found/i"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.formatCommand] = verible-verilog-format"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableCompletionProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableHoverProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableSignatureHelpProvider] = false"
[INFO][2025-11-16 09:26:36] ...lsp/handlers.lua:566	"INFO: INFO: settings[systemverilog.disableLinting] = false"
[ERROR][2025-11-16 09:26:36] ...lsp/handlers.lua:562	"ERROR: [Test] updateIfNotInitialized"
```

The Neovim actually sends two consecutive requests, `Initialized` and `DidChangeConfiguration`, to svlangserver. In svlangserver, the `onDidChangeConfiguration()` is executed right after the Promise of `getSettings()` in `onInitialized()` is not resolved. In the normal workflow, we hope that the `onInitialized()` should be completed before `onDidChangeConfiguration()`, based on how the `updateIfNotInitialized` flag is set by evoking the `updateSettings()`. Due to the asynchronous nature of LSP, the behavior may not be applicable.

The symptom can be easily concluded:
1. The `forceUpdate` flag in `updateSettings()` is not set because the first caller of `updateSettings()` is `onDidChangeConfiguration()`, not `onInitialized()`.
2. In [genutil.ts](https://github.com/imc-trading/svlangserver/blob/master/src/genutils.ts#L112), there is a small typo that the `sortedStringList2` is using `stringList1.sort()`. Therefore, the `isStringListEqual` using by `updateSettings()` is not working. Therefore, the `svindexer.index()` is not called even if the config is changed without the `forceUpdate` flag, unless one of the StringList is undefined.
3. The `setLibraries()` is called without execution of `svindexer.index()`.
4. The `_srcFiles` is not defined because `class SystemVerilogIndexer` did not initialize it. Therefore, `this._srcFiles.map()` will return `TypeError: Cannot read properties of undefined (reading 'map')`

I'll open a PR to optimize this.

closed 2025-11-17 by kkanhere